geekworm-com/x735-v2.5 #7: The terminal is abnormal after restarting the device

350887156, 2021-08-13:
```
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "/home/pi/x735-v2.5/pwm_fan_control.py", line 8, in <module>
    pwm.set_mode(servo, pigpio.OUTPUT)
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1376, in set_mode
    return _u2i(_pigpio_command(self.sl, _PI_CMD_MODES, gpio, mode))
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1025, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

```

### uname -a 

```
Linux raspbian 5.10.42-Release-OPENFANS+20210611-v8 #1 SMP PREEMPT Fri Jun 11 17:19:45 CST 2021 aarch64 GNU/Linux
```

geekworm-com, 2021-11-30:
which OS?
The pwm fan control is depend on the pigpiod liberary.

350887156, 2021-12-05:
@geekworm-com 

Debian-Pi-Aarch64
[https://github.com/openfans-community-offical/Debian-Pi-Aarch64](url)