geekworm-com/x735-v2.5 #10: Rasperry turns off after install

petikeje, 2021-11-29:
I installed the x735 v2.5 to my rasperry pi with bullseye. I can't start the pi after reboot because during the boot it turns off itself with a message "X735 Shutting down..."

geekworm-com, 2021-11-30:
Hi, shut down then turn on

AndBaetz, 2022-08-21:
Mine is not shutting down, but at start up it always shows "X735 Shutting down...". That is kinda confusing, but it seems to come from the script that is started via rc.local (https://wiki.geekworm.com/X735_Software); see the echo here:
[image]

Should that not be like "X735 Shutdown Script started" or something?