geekworm-com/x735-v2.5 #20: install-pwr-service.sh powers Pi off, OS won't boot

vlig, 2024-09-28:
```
$ sudo ./install-pwr-service.sh
Start installing the power management service...
Power management service installed
$
Broadcast message from root@pi41 (Sat 2024-09-28 12:45:57 MSK):

The system will power off now!
```
Doing steps according the manual. Then the OS isn't booting, just shows the logo, then powers off again. Only deleting the /lib/systemd/system/x735-pwr.service file from the microsd card makes the OS bootable again.

P.S. Please, translate Chinese to English in the repo files.

geekworm-com, 2024-09-28:
This project is deprecated, please refer to the new repository https://github.com/geekworm-com/x735-script

vlig, 2024-09-28:
> This project is deprecated, please refer to the new repository https://github.com/geekworm-com/x735-script

Yes, I'm using it certainly, sorry. I've just written to the wrong repo. I'll make my addition in the comments section there, thanks.